neurobooth/neurobooth-os #434: FLIR : AccessException and OutofRange exception thrown randomly during sessions

divyadk, 2024-10-04:
FLIR: Error Setting Value [SetValue(195)]: Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'AcquisitionFrameRate' while calling 'AcquisitionFrameRate.SetValue()' (file 'FloatT.h', line 83) [-2006]

FLIR: Error Setting Value [SetValue(195)]: Spinnaker: GenICam::OutOfRangeException= Value 195.000000 must be smaller than or equal 98.064289. : OutOfRangeException thrown in node 'AcquisitionFrameRate' while calling 'AcquisitionFrameRate.SetValue()' (file 'FloatT.h', line 85) [-2002]

These two issues can be inter-related or pointing at the same issue

(NOTE: Update the script flircam.py main to handle the device args)

Getting this error randomly in the sessions. 
Workarounds:
1. Powercycle the FLIR and retest 
2. Check the cables 
(Check FLIR error messages in the log table after 1 and try 2, check error messages again)
3. Handle the resources locking in the setvalue function in the flir.py

lwhite1, 2024-10-10:
See below for the counts by month for all flir issues since we started logging to the database.  Although we just have a few days of data so far, it looks like rebooting has had an impact. We have had 13 occurrences in the 7 sessions recorded Monday through Wednesday this week, and 44 in the 12 sessions recorded in the first four days of the month 

Year-Month    Count
"2024-03"	27
"2024-04"	18
"2024-05"	20
"2024-06"	32
"2024-07"	40
"2024-08"	81
"2024-09"	104
"2024-10"	57

lwhite1, 2024-10-14:
Followed up with @boubre on his reset code, and why he removed it. For the record, here's what he had to say: 

> That bit of code was really the result of one of those debugging sessions where the issue was not at all clear and we were trying line-by-line changes to see what was going on. So I unfortunately do not have a great grip on what the issue might be.
However - one thing I learned later was the potential for software version mismatches with the FLIR. We have dedicated software for the FLIR installed on ACQ and a python library installed to interface with the software. Both of these are obtained from the manufacturer and it's possible to install different versions for each of these and have things sort-of-work. So I would triple check that these are the same before trying to zero in further on an issue. (edited)

lwhite1, 2024-10-31:
@divyadk Do you have a sense of when you'll be able to check the video files to see if problems setting FPS are impacting the actual FPS of the videos?